bigtreetech/BIQU-B1-SE-PLUS #7: Firmware source files and wifi support

inferior-minion, 2021-09-03:
Hello.

I'm hoping you'll eventually release the source files of the firmware so the community can modify it to suit whatever upgrades we may install on this printer. Currently, I only see pre-compiled firmware.bin files. I'm planning on installing an all-metal hotend at some point so that I can print at higher temperatures so I'll need the source files in order to enable higher hotend temperatures.

I also have a wifi module already: BTT ESP-07S Module
https://www.biqu.equipment/collections/all-product/products/btt-esp-07s

I would like to be able to use this module with the printer, but currently there is no firmware and/or instructions on how to use this module with the printer. From what I can see in the marlin source code we'll need to modify the Configuration_adv.h file by uncommenting #define ESP3D_WIFISUPPORT and then create a file called "Configuration_Secure.h" that looks like the following:
#define WIFI_SSID "WiFi SSID"
#define WIFI_PWD  "WiFi Password"

Again, from what I understand at this point getting the wifi working should be as easy as compiling the new firmware, and uploading firmware.bin and Configuration_Secure.h to the motherboard on this printer to update it.

whoishack, 2021-09-26:
Here for the source files, too!

thisiskeithb, 2021-11-17:
If you're running Marlin, the strain gauge used in the B1 SE's hotend is now fully supported in the latest `bugfix-2.0.x` (and will be in `2.0.9.3`.

Grab the [B1 SE Plus config from the Marlin Configurations repo](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/BIQU/B1%20SE%20Plus) and you'll have a base firmware to customize/enable anything you'd like.

Ejay-3D, 2022-01-30:
hello @thisiskeithb Is the source firmware for SE PLUS IDEX is available somewhere ? Thank you

tinuva, 2022-02-10:
Maybe he meant https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/BIQU/B1%20SE%20Plus ?
https://github.com/MarlinFirmware/Configurations/tree/2.0.9.3/config/examples/BIQU/B1%20SE%20Plus ?