bigtreetech/BIQU-B1-SE-PLUS #12: Biqu B1 SE PLUS marlin configuration files

vakeiros, 2021-12-07:
Hello, I bought a BIQU B1 SE Plus and would like to do some mods to it would be very appreciated if you guys could provide the marlin configuration files for this printer, or at least give out the components used and specs like what thermistors, motor specs, motion specs, etc.

Zedar1968, 2022-02-22:
Hi. Sorry for the late answer but I've only joined the community recently myself. The necessary files can be found here:
https://github.com/MarlinFirmware/Configurations/tree/release-2.0.9.3/config/examples/BIQU/B1%20SE%20Plus

snapo, 2022-06-19:
Problem is with that configuration file you dont get the TFT32 BTT interface... you only have then the marlin interface. How do we get the BTT interface?
Additionally i compiled Marlin 2.1.x with the provided config file and reading the SD card or USB stick was not possible and the board reboots.

vakeiros, 2022-06-19:
you can't. you either use btt or marlin firmware, you can't use both with that screen. I dropped both marlin and BTT firmware on that machine long time ago, I've been using klipper and its like its not the original machine anymore

snapo, 2022-06-19:
i am ok with only the BTT interface, but when i use the configuration it for example:
Marlin 2.1.X
Configuration 2.1.X bugfix

when i try to print something from the sd card it shows the content , but when selecting one, confirmation screen yes, reboot
when i try to print something from the usb stick, it shows the content, selecting the gcode file, confirm screen yes, starts to heat up and reboot.

Do you know by mind how i can activate the BTT interface in the configuration (because i do not want the marlin interface)...
Would greatly appreciate some information on where in Marlin the BTT interface is activated....

Zedar1968, 2022-06-19:
I never use the BTT interface myself so my own config files wouldn't help you plus I have duel Z axis as well as several other mods. The only BTT file download I know of is found here: https://github.com/bigtreetech/BIQU-B1-SE-PLUS/tree/master/B1-SE-PLUS%20firmware/BTT-mode if that helps.

vakeiros, 2022-06-19:
I know nothing about the BTT interface... the little time I used "original" firmware it was mainly marlin and I got sick of it and swapped to klipper. I already had klipper on my voron2.4 so I was used to it, only bought the SI SE cause i got a good deal the Idea was to fully mod it from day1...so no I can't help you with that

snapo, 2022-06-19:
@Zedar1968 that is the full firmware pre-compiled.... i know that one works "kinda ok" still has a lot of bugs.
What i try to achive is running marlin 2.1.X (which i have running currently) only some minor bugs.... But the big problem i really do not know how to solve is, how do i activate the BTT interface in the Configuration_adv.h or Configuration.h file.... 

Because on the B1 SE Plus (new version which i have) it seems the TFT is directly connected to the SKR2 board and the display does not seem to have a separate SD slot.

So the only thing i am struggeling is: getting the BTT interface into the Marlin sources and replace the Marlin interface with the BTT one.

Zedar1968, 2022-06-19:
I believe all you need to do is place the linked files onto an SD card, insert the card into your B1 and turn it on. The files should be recognised as TFT firmware and it should then install. You should then have the btt interface.

snapo, 2022-06-19:
Marlin 2.1.X is now running very smooth...
Finally no more castrated acceleration / speed limit from the stock firmware... :-)

Only bug : when sdcard and usb drive is inserted at the same time the printer reboots every 2-3 minutes. So only use the sd card OR the USB stick.

I think i just leave it like it is and dont give a f... about the BTT interface... Marlin interface just requires more clicks...

Now i start to like the printer, before i was close to putting it in the trash.

Zedar1968, 2022-06-19:
When you compile the firmware it is either compiled for USB or SD one or the other. You are only supposed to use the method the firmware is compiled for.

snapo, 2022-06-19:
I always use the sdcard for firmware flashes... and for printing out of convinience i use the usb stick

Zedar1968, 2022-06-19:
My advice would be to use whichever method the firmware was built for to both flash and print. I always build the USB build of the firmware then flash it from a USB and have never had any problems at all. The other thing with the SE plus is that there are two different control boards and you have to make sure you are flashing the correct version for the board you have. I was talking to someone on a Facebook group who was having strange a reboot problem after updating her firmware and that turned out to be the fact she had flashed the firmware for the wrong board.

Otto7714, 2022-09-22:
Hello, I want to configure the xyz steps, could someone help me how to do it? do i have to install marlin first?

snapo, 2022-09-22:
@Otto7714 normaly you should be able to change them with the gcode directly...
https://marlinfw.org/docs/gcode/M092.html

For example in the slicer you use, you edit (add) the M92 code with the steps that you want for all axis....

Best regards

Otto7714, 2022-09-22:
oh I understand!  thanks... I have a b1 and the steps are configured directly from the printer.  I tried to do it the same way on the SE PLUS and I couldn't find the configuration... it confused me a bit

Keelan-Moore, 2023-02-22:
@snapo Have you managed to get PLR working? 

The precompiled firmware is buggy (marlin mode just rehomes z and restarts the print ontop of your failed print) but the config on Marlin repo doesn't seem to work for me either, the SD card doesn't automatically mount so it doesn't read the power loss file, if its even being written.... who knows.

helakejr, 2023-02-22:
After the flash did you set defaults and save to EEPROM?

Keelan-Moore, 2023-02-23:
> After the flash did you set defaults and save to EEPROM?

If you mean by. defaults you mean activating the PLR feature and saving then yes I did make sure I did that. I've also tried several versions of, all having similar issues (2.0.x, 2.1.2, bugfixes versions for 2.0.x and 2.1.x)

helakejr, 2023-02-23:
M502 then M500.

tp3221, 2023-07-23:
@snapo would you mind sharing your configuration files? I'm still having issues on Marlin 2.1.x and random reboots

helakejr, 2023-07-24:
Here are my Marlin-2.1.2.1.BIQU-B1SE-Plus configuration files.
[Configuration.h.txt](https://github.com/bigtreetech/BIQU-B1-SE-PLUS/files/12140396/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/bigtreetech/BIQU-B1-SE-PLUS/files/12140397/Configuration_adv.h.txt)

tp3221, 2023-07-25:
You're amazing, printer has been going half the day straight with no reboots. Not sure what could have caused the reboots on my compiled firmware.

helakejr, 2023-07-25:
@tp3221 glad it worked out for you.