bigtreetech/BIQU-B1-SE-PLUS #13: Randow Thermal runway E1

Ejay-3D, 2021-12-24:
I'm using a single extrudeur and I'm getting this random message during print . 
[image]
[image]

Machine is flashed with this firmware : https://github.com/bigtreetech/BIQU-B1-SE-PLUS/tree/master/B1-SE-PLUS%20firmware/BTT-mode

Can you help  ? 

Thank you

radek8, 2021-12-28:
Check the tetmistor on the second nozzle

Ejay-3D, 2021-12-28:
I'm using only ONE nozzle which is E0 . Should not be E1 ... E1 doesn't exist on my machine

radek8, 2021-12-28:
Send a response to the M503 and  M115 command

Ejay-3D, 2021-12-28:
Look like it's not just my machine ..... There's an issue somewhere. My firmware is the last one posted in this repo
[image]
[image]
[image]

radek8, 2021-12-28:
@bigtreetech 
This looks like a Marlin configuration error on BIQU-B1-SE-PLUS.

@Ejay-3D 
Compile your own FW. It may take a long time before the BTT fixes it.

radek8, 2021-12-28:
Try connecting a 100Kohm resistor to the connector for Thermistor E1.

radek8, 2021-12-28:
> Odešlete odpověď na příkaz M503 a M115

Ejay-3D, 2021-12-28:
Yes I plan to do that compile my own firmware maybe you can help by answering this question . I have a few B1 and I like all the option you can have on the screen ( eg terminal, Auto PID etc .... )  I have the TFT35 sources and Icon, but it seems that now the TFT is updated directly by the motherboard flash . How is done the relationship between marlin option and the icon .... For instance how to add a new icon or menu that will allow to use terminal directly on the machine ?   To answer you post " _Try connecting a 100Kohm resistor to the connector for Thermistor E1._ " That doesn't make sense since the machine is able to print as this and the error happen randomly , ( like i'm now printing for 4 days without error ) I didn't change anything since last error ...

radek8, 2021-12-28:
You have 2 FW in the printer.
one is on SKR, it's Marlin.
The second is on the display.

The current FW for the display can be found here:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update
if you do not speak C, do not edit the icon layout.

You can find the current FW Marlin here:
https://marlinfw.org/meta/download/

You can see the configuration for B1 here:
https://github.com/bigtreetech/Marlin/tree/B1-2.0.9.2.x

radek8, 2021-12-28:
Send me the answer to M503 and M115 !!!!

radek8, 2021-12-28:
I simulated a mistake with myself.
E1 is the first Extruder.
In FW, it is marked as E0 and the error message says E1.
Then it's good.
There are several ways why you have this mistake.
1. You have a problem with the thermistor. A damaged thermistor, a damaged lead to it or a connector has poor contact.
2. The FW Marlin is set to a high sensitivity to heat leakage.
3. You have a poor heating element and you are printing at a high temperature and the heating element does not have time to heat the nozzle to the required temperature
4. Incorrectly set PID on the nozzle

Ejay-3D, 2021-12-28:
1 __You have a problem with the thermistor. A damaged thermistor, a damaged lead to it or a connector has poor contact._**Nope brand new .... ( I mounted a hundred of head so I can consider i'm experimented )**
2 _The FW Marlin is set to a high sensitivity to heat leakage._ ****Yes probably could be a cause**** 
3 _You have a weak heating element and you are printing at a high temperature and the heating element does not have time to heat the nozzle to the required temperature_  **nope printing at 210 ....** 
4 _Incorrectly set PID on the nozzle .._. **I agree ( that's why i'm looking for adding some option like this in the firmware and screen )**

radek8, 2021-12-28:
The print cooling fan could still blow on the nozzle and cool it too much, but that would be more of a weak heater problem.
Can you measure the output of the nozzle heater?

If PID tampering is enabled in Marlin, then in Marlin mode it is possible and in touch mode there is an icon for it :-)

radek8, 2021-12-28:
We still run into the problem of precompiled FW.
You don't know what values are set, what features are enabled, what HW is supported ....
Without Marli's own compilation, this problem will be difficult to solve.
I don't understand why BTT can't publish a source from precompiled FW ...

Ejay-3D, 2021-12-28:
This is the source I have request and Biqu shared it here ; https://github.com/bigtreetech/BIQU-B1-SE-PLUS/raw/master/B1-SE-PLUS%20firmware/B1-SE-SKR2(2.6.0).rar

radek8, 2021-12-28:
will you open the file? It tells me that the archive is damaged.

Ejay-3D, 2021-12-28:
Archive is not damaged I have just download it and extract it without prob . Send me your mail I can drop it to you if you like .

Ejay-3D, 2021-12-28:
[B1-SE-SKR2(2.6-1.0)-2021-09-24.zip](https://github.com/bigtreetech/BIQU-B1-SE-PLUS/files/7784743/B1-SE-SKR2.2.6-1.0.-2021-09-24.zip)

radek8, 2021-12-28:
WINRAR update helped :-)

radek8, 2021-12-28:
The nozzle thermal protection setting is standard.
The bed setting is too loose.
The LCD PID tuning menu is off.

The submitted source is old, but it is not a problem to compile a new one and adjust its parameters.

radek8, 2021-12-28:
Do you have a PID tuning menu on the touch screen?
if not try it through the terminal.
M303 E0 C8 S210

Ejay-3D, 2021-12-28:
I do not have this in the menu , and the printer is not USB connected .. SO I can't run any command for now ...

radek8, 2021-12-28:
The display has a command line terminal.

radek8, 2021-12-28:
The FW you have on the display is not standard.
It's kind of trimmed and trimmed.
I would start by updating the display to the current versions.

Ejay-3D, 2021-12-28:
Ok can you share a source or a full compiled version with the TFT icons ?

radek8, 2021-12-28:
I have already sent you the link above

radek8, 2021-12-28:
Here's how. Depending on the HW version, select the correct FW and icons.
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

Ejay-3D, 2021-12-28:
B1 SE PLUS TFT doesn't have SD input card so That can't be used .... Or i don't know how ...

radek8, 2021-12-28:
can you take a picture of that screen from behind?

radek8, 2021-12-28:
Did they put such a pathetic display in the printer?

radek8, 2021-12-28:
I thought they put standard TFT displays on offer. Not so crazy.

Ejay-3D, 2021-12-28:
TFT35 SPI 
[image]
[image]

radek8, 2021-12-28:
So that's nice shit :-(

radek8, 2021-12-28:
We can't think much of it with this display.
So try to tune the PID via USB.
And start working on your own Marlin compilation.

Ejay-3D, 2021-12-28:
I agree that's a big fail TFT 35 SPI ... Why ??? @bigtreetech  Your force is to be open source ... But you do not release any docs on the hardware ... I DO NOT WANT to plug my printer to my PC ... So please explain how the hardware works , and the user could eventually improve the software ...  So far no one from BTT looked or answered on this issue ...That's not good at all ... For know even if I want to help I can't because lack of doc ...

Ejay-3D, 2022-01-30:
Closing This issue, The resolution is that the ribbon cable need to be very well fixed to the gantry, so the connector on the head doesn't move and so no cable could be damaged by the head movement . My brown cable ( thermistor  cable) was a bit damaged and so create some random shorts. After reparing the connector and well fix it to the gantry , no more issue .

closed 2022-01-30 by Ejay-3D