bigtreetech/BIQU-B1-SE-PLUS #14: B1 SE Plus - TMC Connection Error

pairoatn, 2021-12-25:
[image]
The issue is TMC connection Error and trying to print and found layer not stick together. What should I do next?

radek8, 2021-12-28:
When did the error appear? After the upgrade?

pairoatn, 2021-12-28:
Error at the beginning that's why customer can't print completed at all.

radek8, 2021-12-28:
There is a problem with the motor controllers.
Send a response to the M122 command

pairoatn, 2021-12-28:
Connecting...
Printer is now online.
>>> M122
SENDING:M122
X Y Z E
Enabled false false false false
Set current 800 800 800 800
RMS current 1436 1436 1436 1436
MAX current 2025 2025 2025 2025
Run current 25/31 25/31 25/31 25/31
Hold current 12/31 12/31 12/31 12/31
CS actual 0/31 0/31 0/31 0/31
PWM scale
vsense 0=.325 0=.325 0=.325 0=.325
stealthChop false false false false
msteps 256 256 256 256
interp false false false false
tstep 0 0 0 0
PWM thresh.
[mm/s]
OT prewarn false false false false
pwm scale sum 0 0 0 0
pwm scale auto 0 0 0 0
pwm offset auto 0 0 0 0
pwm grad auto 0 0 0 0
off time 0 0 0 0
blank time 16 16 16 16
hysteresis
 -end -3 -3 -3 -3
 -start 1 1 1 1
Stallguard thrs
uStep count 0 0 0 0
DRVSTATUS X Y Z E
sg_result
stst * * * *
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
X 0x00:00:00:00 Bad response!
Y 0x00:00:00:00 Bad response!
Z 0x00:00:00:00 Bad response!
E 0x00:00:00:00 Bad response!
Testing X connection... Error: All LOW
Testing Y connection... Error: All LOW
Testing Z connection... Error: All LOW
Testing E connection... Error: All LOW
Received command TMC CONNECTION ERROR
//action:notification TMC CONNECTION ERROR

radek8, 2021-12-28:
Check if there are pins for the UART mod under the motor controllers.

radek8, 2021-12-28:
Check if the motor controllers have a surge for UART mod.

aguinaldoabbj, 2022-08-04:
Any updates on this?