bigtreetech/BIQU-B1-SE-PLUS #15: Z homing dosent work after IDEX upgrade on BIQU B1 SE PLUS

mackna, 2022-01-14:
I have problem with Z homing after I upgraded to IDEX om my BIQU B1 SE PLUS.

I followd all instruction, rechecked my wiring but cant get the Z probe to work. I just keeps going down even after the nozzle touches the bed and the blue led is switch on.
I check Allens video on youtube (https://www.youtube.com/watch?v=hTIuQUC02N0&t=393s&ab_channel=ALLenDouglas) and have the same wiring as him. I flashed both Marlin and RepRap and none of them work. 
I dont know what to to next. I also checked with M119 and all endstops is working execpt Z probe, just reports "open" never triggerd.

radek8, 2022-01-15:
What Marlin board is used in the printer?
Where did you connect the Z probe connector?

mackna, 2022-01-17:
I follwed Allens video on youtube how to connect all cables. It uses SKR 2.

The problem is in the firmware, manage to get it working by recompiling my own firmware. But now i cant get the X2 driver to work, ALL LOW, and bad response is what i get from the M122 command. 

I´m giving up soon, how can BIQU be irresponsible, its embarrising. Missing cables in the box, incomplete manual to upgrade to IDEX, non working firmware for the IDEX, both Marlin and RepRap ( we are many having the same issue)..

radek8, 2022-01-17:
There are two standard BLTouch connections. You must use the one you have configured in Marlin. If one doesn't work try the other.
[image]
[image]

MaCJaX, 2023-01-14:
It's must be the first diagram you posted as the 2nd on which shows the probe on Z-stop is for the X max stop or the new E1 min stop.

MaCJaX, 2023-01-14:
> It's must be the first diagram you posted as the 2nd on which shows the probe on Z-stop is for the X max stop or the new E1 min stop.

MaCJaX, 2023-01-14:
BTW, X2 isn't working because you need to place a jumper in the correct location under the stepper driver. Just copy how the jumpers are on the SKR board under the drivers (all the same). I bet you don't have a jumper anywhere because no where in the manual or video does it show to put one there.