bigtreetech/BIQU-B1-SE-PLUS #19: Error: All Low after boot up

adverant, 2022-09-05:
After booting up with new firmware for the dual IDEX Upgrade kit with dual Z axis I get the error:

All Low
and
Failed to enable Bed Leveling

Homing does not work
[image]

adverant, 2022-09-05:
Here's the screenshot of information screen:
[image]

MaCJaX, 2023-01-14:
I have the same issue. Did you ever fix this ? Thanks !!!

adverant, 2023-01-14:
Not yet.  I ordered a new motherboard that I will install soon.  I’ll keep
you posted if it works.

Don

MaCJaX, 2023-01-14:
This is my 2nd SKR 2 board and same issue. I thought the board too. Have you type in M119 at the terminal prompt to see which end stops are opened and which are triggered, etc ?

adverant, 2023-01-14:
Ouch.

I have not tried that.  I will though - thanks for that.  I’m away from the
printer now but will give it a shot the next time i have a chance.

Don

MaCJaX, 2023-01-14:
No problem. Please also type M122 and post your results here. Thanks !

MaCJaX, 2023-01-14:
Hey, I fixed it ! But it's using E1 to home Z instead of E0. Ugh, god damn BiQu. 

To fix, pull off the TMC driver from the secondary board and place a single jumper in the same position like the ones on your main SKR board. I bet you don't have a jumper there. NO WHERE in ANY video or guide I looked at shows them installing a jumper in this location. Such a waste of time. 

But now, I gotta figure out why E1 is set to probe and not E0. 

Let me know how you make out....

adverant, 2023-01-14:
awesome. nice - i’ll try this.

Don

MaCJaX, 2023-01-14:
Install that jumper and use the firmware in this link. It fixes everything. You will need to compile it with Visual Studio or Arduino IDE. If you can't let me know and ill send you the .bin file. 

https://github.com/mackna/Marlin/tree/SE-Plus-IDEX-2.0.9.3.x

adverant, 2023-01-14:
ok great thanks again :)

Don

Techno11, 2023-01-27:
@MaCJaX would you be able to send your .bin file? I was unable to get it to compile properly.  I had the same LOW issue, and it was resolved by adding the jumper.  However, now I face a crashing/rebooting issue... not sure what causes it but it will randomly crash/reboot about ~1.5 minutes after turning on (or about halfway through the ABL sequence) I'm curious if whatever changes you made to the firmware may have fixed this crashing issue.

Curiously enough, the crashing/rebooting doesn't happen on the first boot after the flash.  If I press the reset button or power cycle the printer after the first boot on a fresh flash cycle, it will start the crashing/rebooting again.  I am running the latest IDEX _seperate_z firmware.

MaCJaX, 2023-01-27:
Hey Techno11. I also had that same rebooting issue and figured out it was due to the SD card. Boot up printer without an SD card and keep it out and see if it still reboots. Unfortunately, I didn’t find a work around. There was some options I read online about disabling a few things in the printers settings (forgot what it was) but I didn’t go that route. I use octoprint so I just now only print with Octoprint and not an Sd card and I don’t experience the reboot. Hope this helps.

Techno11, 2023-01-29:
This seems to be the cause. Thanks! For what it's worth, I tried the most recent build of the B1-SE-Plus from the bigtreetech/Marlin repo and it went back to having the ALL LOW error, so I gave up. I will go back to the release in this repo and just print from USB. Thanks!

MaCJaX, 2023-01-29:
Dont give up ! Theres a reason why you are getting that error. I found the BiQu firmware does work and there were mistakes on my part. How is your 2nd Z motor setup ? Is it connected to the main board or daughter board with a separate stepper motor ?

helakejr, 2023-01-29:
My dual Z motors are connected to ZA and ZB of the SKR2 motherboard. The
dual Z kit also contains another driver to swap with the existing TMC2208
driver. One TMC driver for two Z axis motors.

MaCJaX, 2023-01-29:
Ok so theres 2 different types of firmware on biQus website for the IDEX setup. Make sure you using the one that doesn’t say “dual z axis” as that firmware is for separate stepper drivers and you are using ZA/ZB. This is why you are getting “ALL LOW” error. 

 Let me know …